SeitaBV/v2g-liberty #25: Refactor response error handler

Flix6x, 2022-11-14:
We should refactor so that these functions are defined in one place, for example, as methods of a mixin class, that is subclassed by both this class and `FlexMeasuresClient`.

_Originally posted by @Flix6x in https://github.com/SeitaBV/v2g-liberty/pull/17#discussion_r963115897_

ArdJonker, 2022-11-14:
Good suggestion.
Maybe we could also have FlexMeasuresClient handle all communication with FM. Have FlexMeasuresDataImporter and SetFMdata call a function on FlexMeasuresClient.
This would also be a nice pattern to use for WallboxClient.
What is you opinion?

Flix6x, 2022-11-15:
That is a good thought. Then it would make sense that the response error handler is just a method of the FlexMeasuresClient. That is, we won't need a mixin.

The reponse error handler is not used by the WallboxClient - the handler deals with platform reauthorization, which is not relevant for the modbus communication to the charger - so that would be a different issue.

ArdJonker, 2024-08-08:
Has been picked-up in the addon-v2g-liberty repository.

closed 2024-08-08 by ArdJonker